a2633063/zTC1 #1: Please adapt the A2 version

Landy908, 2019-04-02:
As we all konw,the TC1 has two different versions and you have adapt the A1 version. However,at the same time, the majority of consumers have the A2 version, especially, most of the consumers came from “Big cousin's shop”.For the benefit of our comrades，I kneel to request our author to adapt the A2 version.
Thanks a lot.

Thisyf, 2019-04-02:
Big cousin's shop

linlyv, 2019-04-02:
我也在那个店买了A2版的……

jiafengwang, 2019-04-03:
Big cousin's shop...

iYongHsu, 2019-04-07:
What?我也是在这个店买的，还一下子买了3个。都是A2的吗？我还没拆呢。

a2633063, 2019-04-07:
正在完成dc1固件,完成后开始tc1 a2固件,业余时间完成,不确定完成时间

andyhuai, 2019-04-08:
> What?我也是在这个店买的，还一下子买了3个。都是A2的吗？我还没拆呢。

Big cousin's shop是什么店？ 现在pdd都在70上下的价格

samghb, 2019-04-08:
同问“大表店”是什么店，没看懂。。。

iYongHsu, 2019-04-10:
> > What?我也是在这个店买的，还一下子买了3个。都是A2的吗？我还没拆呢。
> 
> Big cousin's shop是什么店？ 现在pdd都在70上下的价格

就是pdd的一个叫做   大表哥的小店 的店的TC1，69.

laris, 2019-04-11:
today have 65 -5 (new account)

samghb, 2019-04-12:
是不是在这里不讲英文会显得格格不入?🐶

Landy908, 2019-04-12:
> 是不是在这里不讲英文会显得格格不入?🐶

没错，就是你，格格不入【/滑稽/滑稽/滑稽】

jiafengwang, 2019-04-12:
> > 是不是在这里不讲英文会显得格格不入?🐶
> 
> 没错，就是你，格格不入【/滑稽/滑稽/滑稽】

Please speak Yingwen

laris, 2019-04-16:
谁能提供A2的主控照片拆解？
目前还没在网上查到任何详细信息

Who can upload A2 version WIFI-MCU photo ?

a2633063, 2019-04-16:
> 谁能提供A2的主控照片拆解？
> 目前还没在网上查到任何详细信息
> 
> Who can upload A2 version WIFI-MCU photo ?

A2版主控已经确认了

laris, 2019-05-01:
> > 谁能提供A2的主控照片拆解？
> > 目前还没在网上查到任何详细信息
> > Who can upload A2 version WIFI-MCU photo ?
> 
> A2版主控已经确认了

分享一下？

laris, 2019-05-11:
https://github.com/linlyv/TC1_A2
找到了