a2633063/zTC1 #21: 固件BUG

tljim, 2019-04-29:
插口4设置成电脑，周一到周五7:50开，18:00关。插口6设置成手机充电，每天9:30开，11:00关。现在的BUG是，每天9:30手机充电不会打开，反而电脑插口到9:30会自动关闭.希望修复BUG，固件版本0.10.1

tljim, 2019-04-29:
APP版本0.9

tljim, 2019-04-29:
补充：刚才试了下，我把电脑插口的定时关了，手机插口定时就正常使用了

godt2p4, 2019-05-06:
经测试，一次性的打开或关闭设定都没有问题，但是定期（全周七天轮回）会有几率失效……全部改成UDP连接后问题有改善

a2633063, 2019-05-06:
是否有使用hass   如果有使用hass 在hass中可以看到开关的历史开关记录 方便确认开关时间

karzn, 2019-06-04:
我的插口3是电脑，每天可以定时开关，然后我想每天定时开关路由器，但路由器一旦按时关闭后第二天路由器一直未上电，按键灯一直在闪烁，请问这个问题该如何解决呢？？@a2633063

karzn, 2019-06-04:
@a2633063 会不会是因为插排ztc1断网后内部的rtc时间停止了，无法更新，所以到了该打开的时间无法打开？？？

a2633063, 2019-06-04:
目前有确认一点是周末百分百导致定时任务失效

a2633063, 2019-06-04:
> @a2633063 会不会是因为插排ztc1断网后内部的rtc时间停止了，无法更新，所以到了该打开的时间无法打开？？？

断网后 rtc也是在走的  只是可能有误差而已

karzn, 2019-06-04:
@a2633063 断网后RTC任务失败的概率很高，一半以上，很容易复现。现在我重新给插排上电，断网后定时任务就无法启动。

karzn, 2019-06-04:
@a2633063 现在我只能把插排的参数复位，否则路由器一直无法上电，感觉插排对网络的依赖很大。。。插排一到周末就可能失败是不是因为，内部的UTC时间由时间戳转UTC的时分秒“周”年月日时有BUG呢?

a2633063, 2019-06-04:
> @a2633063 现在我只能把插排的参数复位，否则路由器一直无法上电，感觉插排对网络的依赖很大。。。插排一到周末就可能失败是不是因为，内部的UTC时间由时间戳转UTC的时分秒“周”年月日时有BUG呢?

周日   周日设计的是7   结果内部sdk计算的认为周日是0导致的   已经解决  但是这么个小问题懒得release一次 所以后面一起release

karzn, 2019-06-04:
> > @a2633063 现在我只能把插排的参数复位，否则路由器一直无法上电，感觉插排对网络的依赖很大。。。插排一到周末就可能失败是不是因为，内部的UTC时间由时间戳转UTC的时分秒“周”年月日时有BUG呢?
> 
> 周日 周日设计的是7 结果内部sdk计算的认为周日是0导致的 已经解决 但是这么个小问题懒得release一次 所以后面一起release

太好了，可否将这版固件的源码、BIN文件发我一下，要不我老得重置参数，配置也比较麻烦。先谢过了。。。邮箱：1399367429@qq.com

a2633063, 2019-06-04:
> > > @a2633063 现在我只能把插排的参数复位，否则路由器一直无法上电，感觉插排对网络的依赖很大。。。插排一到周末就可能失败是不是因为，内部的UTC时间由时间戳转UTC的时分秒“周”年月日时有BUG呢?
> > 
> > 
> > 周日 周日设计的是7 结果内部sdk计算的认为周日是0导致的 已经解决 但是这么个小问题懒得release一次 所以后面一起release
> 
> 太好了，可否将这版固件的源码、BIN文件发我一下，要不我老得重置参数，配置也比较麻烦。先谢过了。。。邮箱：[1399367429@qq.com](mailto:1399367429@qq.com)

因为别商用 不开源    周日前把版本release出来

karzn, 2019-06-04:
> 因为别商用 不开源 周日前把版本release出来

好的，那就等你的release了，我今天还得重置一下。我的不商用，也是自己用、玩儿、学习。

a2633063, 2019-09-23:
更新v1.0.0固件版本,已经修改配网方式为web配网

closed 2021-09-22 by a2633063