a2633063/zTC1 #33: 当打开某一个强电接口时，导致全部强电接口断开

liyiecho, 2019-06-12:
当打开某一个强电接口时，导致全部强电接口断开
[image]

a2633063, 2019-06-12:
请提供版本信息 app版本 hass接入方法等等足够的信息

liyiecho, 2019-06-12:
固件版本v0.10.3，是通过MQTT服务器接入hass的，不知道还要提供什么信息。还有不知道为什么，每次装hass的机器重新通电后启动hass，都要在ZControl app里面点“重启设备”选项，这样hass才能读取到插排信息

a2633063, 2019-06-12:
> 固件版本v0.10.3，是通过MQTT服务器接入hass的，不知道还要提供什么信息。还有不知道为什么，每次装hass的机器重新通电后启动hass，都要在ZControl app里面点“重启设备”选项，这样hass才能读取到插排信息

是hass的mqtt自动发现 还是手动配置? mqtt服务器和hass一个服务器吗 重启hass服务器是否会导致mqtt服务器重启?

liyiecho, 2019-06-12:
> > 固件版本v0.10.3，是通过MQTT服务器接入hass的，不知道还要提供什么信息。还有不知道为什么，每次装hass的机器重新通电后启动hass，都要在ZControl app里面点“重启设备”选项，这样hass才能读取到插排信息
> 
> 是hass的mqtt自动发现 还是手动配置? mqtt服务器和hass一个服务器吗 重启hass服务器是否会导致mqtt服务器重启?

用的MQTT自动发现，MQTT跟hass都布置到同一机器中，也就是说他们会一起重启，但让hass获取插排信息需要在安卓机ZControl app里面点“重启设备”选项

sam0773, 2019-06-16:
确实。。在群晖docker里弄的ha，在套件里弄的MQTT，ha经常看不到tc1，用zcontrol app重启后就识别了。。。

a2633063, 2019-06-16:
请使用手动配置 hass的MQTT自动发现有bug
后期也会删除自动发现功能

liyiecho, 2019-06-17:
手动配置的也一样需要在安卓机ZControl app里面点“重启设备”选项才能识别到。而且，在ZControl app里面更改MQTT地址后（任意一个不存在的地址，不点“重启设备”选项），不会刷新连接状态（ZControl app依然显示服务器已连接，需要重启app才显示断开），HASS里面MQTT地址保持不变，数据仍然会同步到HASS。

a2633063, 2019-06-19:
> 手动配置的也一样需要在安卓机ZControl app里面点“重启设备”选项才能识别到。而且，在ZControl app里面更改MQTT地址后（任意一个不存在的地址，不点“重启设备”选项），不会刷新连接状态（ZControl app依然显示服务器已连接，需要重启app才显示断开），HASS里面MQTT地址保持不变，数据仍然会同步到HASS。

设计如此,无论排插还是app如果有设计过有效的mqtt服务器修改为其他服务器后必须重启生效. 而且mqtt服务器在app上更新后排插也并不是立刻更新数据的   更新排插的mqtt服务器数据是需要点击设备左上角云图标同步mqtt服务器信息.这个在文档里已经有说明.另外 此问题仅讨论 当打开某一个强电接口时，导致全部强电接口断开的问题  如果有其他的主题问题 请重新开issue  如果全部接口断开的问题无法复现,请先关闭此问题

closed 2019-06-19 by liyiecho