a2633063/zTC1 #34: 0.10.3版本功率问题

sam0773, 2019-06-15:
刷固件过程全部按照教程进行，手机app控制功能正常，但是功率一直显示0w...

a2633063, 2019-06-19:
功率为0为已知问题  旧版本同样概率性存在此问题  暂时无解  持续关注

zhongjiazhai, 2019-07-17:
我这边虽然功率不是零，但是功率不准确呀？ 没有插任何设备就显示2.4w 而且有变动 一会儿2.4w 一会儿3.2w，是我这边刷的有问题吗？

a2633063, 2019-07-17:
本身是有功耗的

hwj180, 2019-09-25:
我这有个插排功率正常，有一个插排是刚开始显示为0，过一段时间会显示一个很大的数，1400多w，不知道是硬件原因还是软件原因。

iYJGM, 2019-09-28:
升级后功率为0，是软件原因还是？

CCZU1623, 2019-10-01:
更新v1.0.0版本后出现功率始终为0的问题，之前用的版本没有出现..

CCZU1623, 2019-10-01:
已验证，可能是固件bug，v1.0.0版本过一会儿就恢复功率显示了

a2633063, 2020-05-12:
有人确认功率问题为硬件电路设计影响. 确认更换后可以恢复正常. 
由于手上没有强电的调试设备,此问题应该padding
硬件恢复方法:
https://www.right.com.cn/forum/thread-951885-1-1.html

a2633063, 2020-05-19:
后面的版本将此类问题的功率显示为-1W

shineanr, 2020-05-24:
昨天冒险测试了下芯片的电压 ，没有问题 ，但是功率不显示 ，这个问题确实很郁闷 ，最近才出现的感觉
> 有人确认功率问题为硬件电路设计影响. 确认更换后可以恢复正常.
> 由于手上没有强电的调试设备,此问题应该padding
> 硬件恢复方法:
> https://www.right.com.cn/forum/thread-951885-1-1.html

a2633063, 2020-05-26:
不仅仅是电压  需要确认wifi模组的15pin是否有脉冲信号输入
[image]

正常情况下是有脉冲输入的  测量此信号需要使用示波器或逻辑分析仪测量.

shineanr, 2020-05-27:
是不是这个15脚是接受功率计算芯片的信号的？不懂硬件啊，主要是这个功率芯片不知道是不是坏了，如果是设计缺陷导致坏了，感觉换了以后还是会坏。
> 不仅仅是电压 需要确认wifi模组的15pin是否有脉冲信号输入
> [image]
> 
> 正常情况下是有脉冲输入的 测量此信号需要使用示波器或逻辑分析仪测量.

shineanr, 2020-10-02:
[image]
今天换了下功率芯片确实能显示功率了，希望这个芯片能坚持的久点，又有个一个问题是待机功率3W， @a2633063 不知道后面是否还更新固件，如果更新能否加个校准功率的功能。可以以待机时刻的值为0W为基准计算功率。

shineanr, 2020-10-02:
DC1 是用劫持的方式，功率显示还算准确，TC1是接在DC1上面的。
[image]

a2633063, 2020-10-02:
> [image]
> 今天换了下功率芯片确实能显示功率了，希望这个芯片能坚持的久点，又有个一个问题是待机功率3W， @a2633063 不知道后面是否还更新固件，如果更新能否加个校准功率的功能。可以以待机时刻的值为0W为基准计算功率。

待机功率的3w是包含继电器及mcu的耗电功率的,即功率为排插总功率,并非用电器功率. 也因此不会增加校准功能

shineanr, 2020-10-02:
> 待机功率的3w是包含继电器及mcu的耗电功率的,即功率为排插总功率,并非用电器功率. 也因此不会增加校准功能
这个3w是所有插槽关闭状态下的，如果开启全部插槽，没有任何负载，功率会高达6w，对比另一个全部关闭时0.几w，开启时3w左右，稍微偏差有些大。
[image]

closed 2020-12-29 by a2633063