a2633063/zTC1 #39: 死活连不上，可不可以提供一个默认可以当普通插排的固件？

smart-z, 2019-07-21:
点击配网，要么就是灯常亮毫无闪烁的变化；要么就是会闪烁个不停，然后就没有然后了。

现在一直配网不成功。

能不能提供一个设置或修改，默认可以当作普通插排来使用的机器？

现在这样，插上电除了USB其他的插孔完全不能用，真是太糟心了。请务必解决一下。

a2633063, 2019-07-22:
不提供 
进入配网模式时灯是常亮 如果点配对后边为闪烁证明数据传输了  闪烁表示正在连接wifi 
如果多次尝试还是失败,可以在https://www.mxchip.com/easylink/ 下载app配对试试
另外 配网是有人给我反应容易失败,大部分是**不认真看文档**,少部分路由器固件导致

smart-z, 2019-07-22:
> 不提供
> 进入配网模式时灯是常亮如果点配对后边为闪烁证明数据传输了闪烁表示正在连接wifi 
> 如果多次尝试还是失败,可以在https://www.mxchip.com/easylink/下载app配对试试
> 另外配网是有人给我反应容易失败,大部分是**不认真看文档** ,少部分路由器固件导致

太好了，总算有有点希望了，我折腾折腾 easylink 看看，昨天就是没有找到是那个easylink。先谢了。

LuckyIV, 2019-07-23:
我一开始也是死活连不上，后来尝试把路由器设置里的突发传输（TX优先），智能抗干扰和智能合频关掉后可以连接，具体是哪个选项导致的问题我没有试

ZlWcc, 2019-07-26:
我也配对不了。不同的是灯一直常亮，App一直在“正在配对，请稍后……”。文档都能背了，然而……

ZlWcc, 2019-07-26:
> 我一开始也是死活连不上，后来尝试把路由器设置里的突发传输（TX优先），智能抗干扰和智能合频关掉后可以连接，具体是哪个选项导致的问题我没有试

路由器问题配对不上的情况，点配对LED灯会闪吗？

a2633063, 2019-07-26:
> > 我一开始也是死活连不上，后来尝试把路由器设置里的突发传输（TX优先），智能抗干扰和智能合频关掉后可以连接，具体是哪个选项导致的问题我没有试
> 
> 路由器问题配对不上的情况，点配对LED灯会闪吗？

进入配网模式时灯是常亮 一直常亮可以在https://www.mxchip.com/easylink/ 下载app配对试试

925934787, 2019-07-26:
> > > 我一开始也是死活连不上，后来尝试把路由器设置里的突发传输（TX优先），智能抗干扰和智能合频关掉后可以连接，具体是哪个选项导致的问题我没有试
> > 
> > 
> > 路由器问题配对不上的情况，点配对LED灯会闪吗？
> 
> 进入配网模式时灯是常亮 一直常亮可以在https://www.mxchip.com/easylink/ 下载app配对试试

我重刷了固件，然后现在进入配对模式（断电长按logo按钮），灯闪烁两次之后，常亮，但是常亮只有3-5秒就直接灭了，手机也是一直正在连接，最后很久之后失败，请问一下是不是没刷好，还是插排损坏了？

closed 2019-09-10 by a2633063

nsb2006, 2020-06-16:
> 我一开始也是死活连不上，后来尝试把路由器设置里的突发传输（TX优先），智能抗干扰和智能合频关掉后可以连接，具体是哪个选项导致的问题我没有试

感谢分享，我的德业除湿机在家里换了使用梅林固件的网件路由器之后始终连不上手机，研究了一番查出来德业除湿机用的无线方案是MXCHIP的，一路找到这里，关了路由器2.4G的Tx Burst选项之后成功连接。