a2633063/zTC1 #42: 闭合插口6，突然全部接口都崩了。。。

mzfe, 2019-08-15:
固件版本:0.10.4

App版本:0.11.5

homeassistant  0.97.2

问题描述(包含复现issue方式):
tc1已经接入hass，从昨晚用到现在无任何故障。
插口状态 
1、开  负载中
2、开  负载中
3、关  无负载
4、开  负载中
5、开  负载中
6、开  负载中，接音箱
总功率约为100W，平稳

现象，在 hass 页面闭合6口，试图关音箱
突然，所有接口都关了，logo灯也不亮了（所有强电接口断开）。。。

打开APP看，上次开机时间、运行时间都是正常的，说明wifi模块应该没断，只是强电掉了。

翻看issues，有人提交过 打开强电接口，导致其他断开的问题。而我这里是 关闭强电接口，导致的问题。

mzfe, 2019-08-15:
关键日志：
下午11:20   XX1  turned on
下午11:19   TC1插座 turned off
下午11:19   XX5  turned off
下午11:19   XX4  turned off
下午11:19   XX2  turned off
下午11:19   XX1  turned off
下午11:19  音箱 turned on
下午11:19  音箱 turned off

925934787, 2019-08-21:
我也遇到这个问题，偶发性，频率很高（基本都能复现），关闭开关6，然后就全部都关了

CCZU1623, 2019-08-21:
同样遇到该问题，固件版本0.10.3和0.10.4都有该问题，随机发生，频率较高

a2633063, 2019-08-22:
之前有调试此现象未复现问题
后期了解为必须在有功率的情况下,后面有时间插上设备后再调试此问题

13653081992, 2019-09-24:
> 我也遇到这个问题，偶发性，频率很高（基本都能复现），关闭开关6，然后就全部都关了

我也有这情况我的是开关3有问题

a2633063, 2019-09-24:
我一直无法复现此问题  后续出现此问题的麻烦尽量提供足够的信息
如排插上插了什么设备 功率多少 是否有使用服务器 是app还是hass控制时出现的问题
尽量提供多的信息

13653081992, 2019-09-27:
> 我一直无法复现此问题后续出现此问题的麻烦尝试提供足够的信息
> 如排插上插了什么设备功率多少是否有使用服务器是app还是hass控制时出现的问题
> 设法提供多的信息

我的大概170瓦左右，接了音响，小爱同学，笔记本电源，显示器跟软路由，还有一个插口没用，mqtt是用自己搭的emqx ，一关插口三就全断了

mzfe, 2019-09-30:
> 
> 
> 我一直无法复现此问题 后续出现此问题的麻烦尽量提供足够的信息
> 如排插上插了什么设备 功率多少 是否有使用服务器 是app还是hass控制时出现的问题
> 尽量提供多的信息

感谢作者的回复，自从刚刷好那天出现这个问题后，现在都挺好的。
估计是我闭合开关的次数少吧，也暂时无法重现此问题。
极有可能是仅出现一次的问题 (*^▽^*)

liyiecho, 2019-10-07:
今天再次遇到打开某插口，全部插口断电的情况。这回固件版本是v1.0.0。

liyiecho, 2019-10-08:
@a2633063 今天这情况更诡异。打开插口6，全部插口断电，按下电源键，来回得来电断电，再次按下电源键才正常。固件版本是v1.0.0。
[image]

liangtailin, 2019-12-19:
固件：十月份刷的当时最新的，hass配置与之前不同的那版，应该是1.0.2
使用：1接口UPS接路由器+群晖大概60W，6接口监控电源大概5W，其他接口关闭。
现象：iOS设备定位判断离家，hass触发6号口监控开机（这里我是发两遍相同的turn_on指令，中间间隔一次hass的notify，说明的原因是，不确定是不是第一遍导致所有接口都关闭第二遍才打开接口6），导致1号口UPS断电。
频率：目前每天进出家大概2-3次，每周大概出现此现象2-3次。
目前解决方案：hass间隔3分钟发送一遍1号口打开，非常原始的解决方法，还好有ups。
其他：另外两个插排0.10.3手动测试没有这个问题，不排除没触发问题。

523266656, 2020-03-06:
固件:V1.0.2
负载功率: 15W
总是通过UDP发送数据：是

使用 Home Assistant MQTT 接入 

同样问题 MQTT 控制 开启或关闭 插口 经常全部开关同时关闭 logo熄灭 
MQTT.fx 中监控数据：

06-03-2020 14:04:55.50695873 发送 打开 插口1
device/ztc1/d0bae46376fc/set
`{
  "mac" : "d0bae46376fc",
  "plug_0" : {
    "on" : 1
  }
}`

06-03-2020 14:04:56.50696066 反馈 插口1 已打开
device/ztc1/d0bae46376fc/state 
`{
  "mac" : "d0bae46376fc",
  "plug_0" : {
    "on" : 1
  },
  "plug_1" : {
    "on" : 0
  },
  "plug_2" : {
    "on" : 0
  },
  "plug_3" : {
    "on" : 0
  },
  "plug_4" : {
    "on" : 0
  },
  "plug_5" : {
    "on" : 0
  },
  "name" : "客厅"
}`

06-03-2020 14:04:56.50696236 发送 打开插口2
device/ztc1/d0bae46376fc/set 
`{
  "mac" : "d0bae46376fc",
  "plug_1" : {
    "on" : 1
  }
}
`

06-03-2020 14:04:56.50696539 发送 打开插口3
device/ztc1/d0bae46376fc/set 
`{
  "mac" : "d0bae46376fc",
  "plug_2" : {
    "on" : 1
  }
}
`

06-03-2020 14:04:56.50696625
device/ztc1/d0bae46376fc/state 反馈 插口2 已打开
`
{
  "mac" : "d0bae46376fc",
  "plug_0" : {
    "on" : 1
  },
  "plug_1" : {
    "on" : 1
  },
  "plug_2" : {
    "on" : 0
  },
  "plug_3" : {
    "on" : 0
  },
  "plug_4" : {
    "on" : 0
  },
  "plug_5" : {
    "on" : 0
  },
  "name" : "客厅"
}
`

06-03-2020 14:04:57.50697354
device/ztc1/d0bae46376fc/state 反馈 插口3 已打开
`
{
  "mac" : "d0bae46376fc",
  "plug_0" : {
    "on" : 1
  },
  "plug_1" : {
    "on" : 1
  },
  "plug_2" : {
    "on" : 1
  },
  "plug_3" : {
    "on" : 0
  },
  "plug_4" : {
    "on" : 0
  },
  "plug_5" : {
    "on" : 0
  },
  "name" : "客厅"
}
`

06-03-2020 14:04:58.50698461 发送 打开插口5
device/ztc1/d0bae46376fc/set
`
{
  "mac" : "d0bae46376fc",
  "plug_4" : {
    "on" : 1
  }
}
`

06-03-2020 14:04:58.50698655 反馈 插口5 已打开
device/ztc1/d0bae46376fc/state 
`
{
  "mac" : "d0bae46376fc",
  "plug_0" : {
    "on" : 1
  },
  "plug_1" : {
    "on" : 1
  },
  "plug_2" : {
    "on" : 1
  },
  "plug_3" : {
    "on" : 0
  },
  "plug_4" : {
    "on" : 1
  },
  "plug_5" : {
    "on" : 0
  },
  "name" : "客厅"
}
`

06-03-2020 14:04:59.50699306 后馈  所有插口关闭 
device/ztc1/d0bae46376fc/state
`
{
  "mac" : "d0bae46376fc",
  "plug_0" : {
    "on" : 0
  },
  "plug_1" : {
    "on" : 0
  },
  "plug_2" : {
    "on" : 0
  },
  "plug_3" : {
    "on" : 0
  },
  "plug_4" : {
    "on" : 0
  },
  "plug_5" : {
    "on" : 0
  },
  "name" : "客厅"
}
`

距离上一个反馈 1秒 中间没有发送关闭命令。开或关闭均有一定几率发生此问题

Ilmiotesoro, 2020-04-09:
同样的问题，关闭6号口全部都断开

weiqin1025, 2020-07-13:
同样的问题，使用app，并没有使用ha。
固件版本，1.1.0
app版本，1.4.3
使用情况：
3号口接鱼缸泵6w左右
6号口电风扇35w左右，
其它接通无负载
当手动关闭6号口时，有一定机率关闭所有插口。定时关闭则无此现象发生。

terrance-liang, 2021-02-19:
> 固件：十月份刷的当时最新的，hass 配置与之前不同的那版，应该是 1.0.2
> 使用：1 接口 UPS 接路由器 + 群晖大概 60W，6 接口监控电源大概 5W，其他接口关闭。
> 现象：iOS 设备定位判断离家，hass 触发 6 号口监控开机（这里我是发两遍相同的 turn_on 指令，中间间隔一次 hass 的 [...]

当前时间最新推送固件版本（目前不在家无法连接插排查看版本号，应该是1.2.0），始终勾选“总是通过UDP发送数据”。上述问题仍然存在。

a2633063, 2021-02-19:
已经确认问题点 后续更新固件解决.
当前可以使用打开童锁功能来暂时屏蔽此bug

a2633063, 2021-03-15:
固件1.2.1修复此问题

closed 2021-03-15 by a2633063