a2633063/zTC1 #43: 断网之后强电口全掉

hwj180, 2019-09-09:
请先搜索确认没有重复后再提交issue

固件版本:0.10.3

App版本:

Android版本:

问题描述(包含复现issue方式):断网或者网络不通，强电口全掉，这种思路不合理，如果它的智能部分（脑子）坏掉了，至少能当个普通插排用吧，好比电动车没电了，至少还要留一个脚踏板，当个普通自行陈凑活着用，总不能下来推着走吧。之前无意中不知道操作什么了，强电口全掉，导致插在强电口的路由器关机了，好了，死循环了，路由器现在只能插在墙壁独立插口了。
昨天一个TC1不知道怎么回事，现在只有USB口有电，手机APP里面查看不了状态，路由器里也看不到它的IP地址，按那个状态指示按钮没反应，很可能是wifi模块烧坏了。

a2633063, 2019-09-10:
断网或者网络不通，强电口不会掉啊   
断网的情况下按按钮是可以直接控制所有开关的   如果不行 就是bug了

hwj180, 2019-09-10:
这种现象我碰到两次，就是自己装逼在homeasistant里面来回切换几次之后出现的。最好还是无法连网时或者未激活时让其全部恢复到带电状态，无非就是没法控制了，还可以当普通插排用。

a2633063, 2019-09-10:
homeasistant里面来回切换几次导致断网吗?

hwj180, 2019-09-10:
插排强电口全掉了，然后网络也断了，不知道谁先谁后。
断网的情况下怎么按按钮啊，APP里都不能显示插排的状态了，更不能控制了。
另外，我现在插排的受电插头插上电，按那个phicomm的按键长时间都没任何反应，是不是BOOM了

a2633063, 2019-09-10:
还没有人给我这个反馈.
看起来是之前都正常,之后在hass控制中突然断电断wifi,之后无法启动吗
如果wifi无连接,尝试配网.
如果配网灯不亮,尝试执行恢复默认设置
如果恢复默认设置无反应,看起来可能是固件损坏后硬件损坏了 
能否提供下QQ号码 从QQ上联系比较快

hwj180, 2019-09-10:
278430631

nxcjs, 2019-10-19:
初始能不能取消只能通3个强电的限制，我路由器的电源，只能插在不通的那里，也是死循环

a2633063, 2019-10-20:
> 初始能不能取消只能通3个强电的限制，我路由器的电源，只能插在不通的那里，也是死循环

不激活通3个电源已经是我退一步采取的措施了  本身是不激活不能开的

closed 2019-11-13 by a2633063