a2633063/zTC1 #45: OTA升级，废了

hwj180, 2019-09-24:
> 请先搜索确认没有重复后再提交issue,以下内容若未按照模板填写,不做任何回复,3天后若未更新直接关闭issue.

固件版本(必填):0.10.4 升级1.0

App版本(与app控制有关时必填): 新版APP

Android版本(与app控制有关时必填): ANDROID 9, EMUI 9.1

是否使用服务器(必填):
使用自己的MQTT服务器
问题描述(包含复现issue方式):
手机在外面（非局域网），通过APP能正常显示开关状态，用的是新的APP，勾选了里面的使用旧版通信协议，然后看到有固件更新，点了一下，显示1分钟左右能完成，然后等好长时间，刷新，无法更新开关状态。回家发现开关插口全掉，指示灯不亮，按logo重插无反应。上次好像也是因为手贱点OTA升级崩掉的，OTA升级还是不稳定。上次那个拆开重刷，接JLINK指示灯有反应，能听到继电器动作的声音，但是刷玩还是起不来。估计跟上次一样，砖了，跪求救砖方法，已有两个砖头了。

a2633063, 2019-09-24:
出现此类问题请确认是否还有激活    发现部分出现ota后从已激活变为未激活状态,此为bug
出现无法控制的尝试长按10秒恢复默认设置,如果灯能够亮恢复默认设置则固件正常,确认手法操作是否正确

后续会修改为未激活的就全部打开插口,防止出现此问题

hwj180, 2019-09-24:
长按10s毫无反应，怎么办

925934787, 2019-09-27:
> 出现此类问题请确认是否还有激活 发现部分出现ota后从已激活变为未激活状态,此为bug
> 出现无法控制的尝试长按10秒恢复默认设置,如果灯能够亮恢复默认设置则固件正常,确认手法操作是否正确
> 
> 后续会修改为未激活的就全部打开插口,防止出现此问题

原来ota之后变成未激活是bug啊，还是以为就是这样的，升级之后就未激活了，之前也有过几次莫名其妙变成未激活，手动再次激活就行。

closed 2019-10-08 by a2633063