a2633063/zTC1 #47: nmap扫描插排地址导致插排重启

everything411, 2019-10-07:
固件版本(必填):v1.0.0

App版本(与app控制有关时必填):

Android版本(与app控制有关时必填):

是否使用服务器(必填):
openwrt mosquito mqtt server

问题描述(包含复现issue方式):

无聊尝试nmap扫描插排地址，结果nmap运行过程中，插排断电重启

复现：对插排ip使用nmap必复现 `nmap 192.168.1.105`

如果作者大大觉得这问题太无聊没有修复的价值就Close吧）

膜大佬给设备开发固件 tql

a2633063, 2019-10-08:
没用过nmap,能否提供软件以及说明给我尝试复现下,如果有必要也是考虑修复的...
最好是windows下的

everything411, 2019-10-08:
https://nmap.org/download.html

有windows版本可用，但是我后来发现只有在我的openwrt路由器（也是插排连接的路由器）上执行命令才会这样，从Windows或者linux电脑设备上对其执行nmap不会导致其重启。

猜测可能收到来自网关的某种数据包会重启？还是说Openwrt上移植的nmap与其它系统的实现有一定差异？

不过这样看来好像没什么修复价值就是了:)

```
nmap 192.168.1.105 
# 其中192.168.1.105是插排ip
```

```
root@OpenWrt:~# nmap -v 192.168.1.105
Starting Nmap 7.70 ( https://nmap.org ) at 2019-10-08 05:12 UTC
Initiating ARP Ping Scan at 05:12
Scanning 192.168.1.105 [1 port]
Completed ARP Ping Scan at 05:12, 0.25s elapsed (1 total hosts)
Initiating Parallel DNS resolution of 1 host. at 05:12
Completed Parallel DNS resolution of 1 host. at 05:12, 0.00s elapsed
Initiating SYN Stealth Scan at 05:12
Scanning zTC1_e21e.lan (192.168.1.105) [1000 ports]
Connection reset by 192.168.1.1 port 22
```

liyiecho, 2019-10-08:
2台均已复现，固件版本v1.0.2。

closed 2019-11-13 by a2633063