a2633063/zTC1 #59: 如何主动获取排插整体状态

titianqx, 2019-12-05:
> 请先搜索确认没有重复后再提交issue,以下内容若未按照模板填写,不做任何回复,3天后若未更新直接关闭issue.

固件版本(必填):1.0.2

App版本(与app控制有关时必填):

Android版本(与app控制有关时必填):

是否使用服务器(必填):是

问题描述(包含复现issue方式):

想问下如何主动查询排插整体状态  wiki里没有找到  只找到了被动接受这些信息的

还有就是 设置时 value为null 会返回此key的状态...  

于是我
`const info = {name,mac,plug_0:{on:null},plug_1:{on:null}}`
mqtt的message是
`{topic: 'device/ztc1/${mac}/set',
    payload: Buffer.from(JSON.stringify(info))}`
但请求后 返回来的并不是想要的排插插口状态
看了下wiki里写的 topic为   device/ztc1/MAC/state  是zTC1反馈的数据
但是把message的topic设为这个还是不行

所以想问下 是否存在一个方法，当我发送请求的时候，mqtt会返回排插的整体状态；如果有发送请求时需要发送什么数据。

最后 感谢大佬的无私奉献...

a2633063, 2019-12-05:
mac为唯一标识  如:要获取mac地址为123456789012的排插插口状态,
发送的topic:`device/ztc1/123456789012/set` 内容为:`{"mac":"123456789012","plug_0":{"on":null}}`
排插会在`topic为 device/ztc1/123456789012/state`的topic下回复对应排插状态

titianqx, 2019-12-06:
👌  感谢   我试一下

closed 2019-12-06 by titianqx