a2633063/zTC1 #60: 插排会有不定时断开，配网灯闪动的问题，请帮忙看下

smart-z, 2019-12-18:
我是个小白，有3台TC1，虽然经过不同的情况到最后也都刷了您的固件。
目前遇到一个很重要的问题，您帮忙看看是啥问题？之前用的好好的，最近一个礼拜左右，三台中只有一台会发生插排会有不定时断开，配网灯闪动的问题，故障间隔有可能半个小时，有可能会十几个小时左右发生一次。

- 固件版本(必填):1.0.2
- App版本(与app控制有关时必填):我用pc台式机装了android，装了 SmartControl_Android_MQTT 0.12.5
- Android版本(与app控制有关时必填):  Android 7
- 是否使用服务器(必填):不太懂

我有用MQTT接入 Home assistant。

请帮忙分析分析大概是啥问题，有啥突破口

a2633063, 2019-12-18:
配网灯闪动是否有热点? 是否是恢复成默认设置了?

smart-z, 2019-12-18:
> 配网灯闪动是否有热点? 是否是恢复成默认设置了?

您说的这两个我下次留意一下，看看插排自己是否会开新热点。
这几天我有初始化并重新接入过一次。还是如此。之前却没问题，稳定性还不错。

插排断电后，灯会闪动，重新按灯哪里的按钮，插排会重启，会接入Wi-Fi。因为我的路由器就是在这个插排上，一断电路由器就停了。我把路由单独弄出来看看情况。

smart-z, 2019-12-22:
> 配网灯闪动是否有热点? 是否是恢复成默认设置了?

您好，我这几天又重置并还认真观察了下。之前6个插孔都是满的，把6个插孔里面的路由器和NAS单独弄出来弄到别的插孔了。再减去一个不用的，现在6个插孔还剩三个。
- 之前6个插孔的时候，会不定时偶然断开断电，但配网灯在闪。我仔细看过了，配网等闪的时候没有热点，因为插在他上面的路由器断了，所以他可能在等待配网吧。
- 之后，也会偶然断开。但配网灯却没有之前那样的情况，配网灯就是灭的。好这里打住。
我在仔细查之后的问题原因，发现了这样的情况。我有一台风扇，三角插头插在了tc1的6口上。tc1通过mqtt接入了ha。此外此tc1还3口上插了空气净化器，5口上插了电热毯。我通过ha来控制开关的时候，**只有6插口的电风扇关闭的时候会_偶尔_把全部6个口的开关都关掉**。主要就是这一句，您有没有遇到过类似的情况，我现在还继续观察看看。

a2633063, 2019-12-22:
实际上开关5或6插口会导致所有插口关闭 此问题已经有人和我说过  但是我一直没有复现
未找到问题点 如果能找到百分百复现手法请告诉我.
此问题关注中

smart-z, 2019-12-22:
> 实际上开关5或6插口会导致所有插口关闭 此问题已经有人和我说过 但是我一直没有复现
> 未找到问题点 如果能找到百分百复现手法请告诉我.
> 此问题关注中

我不是程序员也不懂程序，但我知道这玩意如果有日志纪录或者bug调试模式啥的估计可能会有些帮助。现在我的6插口没插东西，这几天再看看5，6插口的情况。

mzfe, 2020-03-15:
> 
> 
> > 实际上开关5或6插口会导致所有插口关闭 此问题已经有人和我说过 但是我一直没有复现
> > 未找到问题点 如果能找到百分百复现手法请告诉我.
> > 此问题关注中
> 
> 我不是程序员也不懂程序，但我知道这玩意如果有日志纪录或者bug调试模式啥的估计可能会有些帮助。现在我的6插口没插东西，这几天再看看5，6插口的情况。
> 
> 再之前的话一直很好，稳定的还不错。也就是我有一次调整了设备插口位置，把1口的三头电风扇，换到6口之后，大概这个时间出现了问题。目前我的5口是二头，6口的三头电风扇换到了4口。说不定跟三角插头有关系？

不是三角插头，我的是两脚插也会。我现在尝试6口接常开的电器了。