a2633063/zTC1 #63: zTC1长时间必定段炼

walk728, 2020-03-01:
> 请先搜索确认没有重复后再提交issue,以下内容若未按照模板填写,不做任何回复,3天后若未更新直接关闭issue.

固件版本(必填):1.0.2

App版本(与app控制有关时必填):

Android版本(与app控制有关时必填):

是否使用服务器(必填):使用mqtt服务器

问题描述(包含复现issue方式):拔TC1的插头重新插入通电，一切正常，但是长时间以后，长时间可能等于几天，app通信断联，（具体为功率无法获取，固件版本号无法获取，自定义的插口名称变成默认的）必须重新拔插tc1的插头才行，按盒子前面的按钮通断不行。拔插以后一切正常。

fantasyjqc, 2020-03-07:
我也遇到这样的问题，一般几天时间会出现一次，mqtt连不上，使用软件udp通信可以控制，必须要彻底关闭插排重启

a2633063, 2020-03-07:
能否在mqtt服务器中确认 排插是否有连上 如果有连上 是否有通信内容?

fantasyjqc, 2020-03-07:
> 能否在mqtt服务器中确认 排插是否有连上 如果有连上 是否有通信内容?

这个只能等到下次出现问题再确认了，我这边使用时间大约1个月，出现过2次这样的情况

walk728, 2020-03-12:
上次发帖的时候重启的，今天出现这种情况，昨天没有看，所以应该是10-11天。如何在mqtt服务器里面看？

walk728, 2020-03-12:
我在ha里面确认了一下，持续运行时间是10天1小时30分钟。然后就失联了。我不知道怎么去mqtt服务器里面查看，可以告诉我一下方法，我去查看

fantasyjqc, 2020-03-13:
> 能否在mqtt服务器中确认 排插是否有连上 如果有连上 是否有通信内容?

今天再次发现出现问题，具体表现为无法设备不再上报插孔开关情况，查看mqtt发现是有连上服务器的，如图：

[image]

但是始终不上报插孔的开关状态：

[image]

与楼上相同，zm1没有任何问题，所以应该不是mqtt的问题

fantasyjqc, 2020-03-31:
> 能否在mqtt服务器中确认 排插是否有连上 如果有连上 是否有通信内容?

今天再次发现出现了这个情况，因为昨天调整家里的网络，断网了一段时间，TC1似乎就不再连接MQTT了，查看MQTT服务器中已经找不到TC1的clinet了。

[image]

walk728, 2020-03-31:
我也一直还是有这个问题，基本上都是10天一次

a2633063, 2020-05-12:
现在问题依然存在吗  mqtt服务器是有自动重连的  我继续关注

walk728, 2020-05-14:
是的，依然存在，10天不操作就断联

a2633063, 2020-07-19:
依然未复现此问题

jjss520, 2020-09-30:
是的，过段时间必定连不上MQTT，mqtt服务器看了，没连上，有点郁闷呀，正好需要远程的时候，打开APP一直，不行，希望能修复这个致命的bug

a2633063, 2020-09-30:
> 是的，过段时间必定连不上MQTT，mqtt服务器看了，没连上，有点郁闷呀，正好需要远程的时候，打开APP一直，不行，希望能修复这个致命的bug
请确认mqtt服务器正常.当前多人使用联系几个月不会断连

stone2289, 2021-01-10:
我最近也出现这种情况了，有时候一两天就连不到mqtt，软件能获取公率/时间，无法获取激活信息，mac地址啥的，也不能控制开关，重新插拔插头恢复正常。固件是v1.2.0，app：v1.10.1，手机是安卓10。

fengsuxing, 2021-01-11:
> 我最近也出现这种情况了，有时候一两天就连不到mqtt，软件能获取公率/时间，无法获取激活信息，mac地址啥的，也不能控制开关，重新插拔插头恢复正常。固件是v1.2.0，app：v1.10.1，手机是安卓10。

和你一样，有功率时间，无法获取开关状态。软件固件都是最新版

fengsuxing, 2021-01-11:
> 我最近也出现这种情况了，有时候一两天就连不到mqtt，软件能获取公率/时间，无法获取激活信息，mac地址啥的，也不能控制开关，重新插拔插头恢复正常。固件是v1.2.0，app：v1.10.1，手机是安卓10。

我和你类似的问题解决了，取消勾选总是UDP发送数据。（我是之前调试自己服务器，临时打开的，忘记关了，远程的情况下就会出现你这种现象）

stone2289, 2021-01-11:
> > 我最近也出现这种情况了，有时候一两天就连不到mqtt，软件能获取公率/时间，无法获取激活信息，mac地址啥的，也不能控制开关，重新插拔插头恢复正常。固件是v1.2.0，app：v1.10.1，手机是安卓10。
> 
> 我和你类似的问题解决了，取消勾选总是UDP发送数据。（我是之前调试自己服务器，临时打开的，忘记关了，远程的情况下就会出现你这种现象）

还不太一样，之前没勾选udp也 出现过，在局域网内app也获取不到信息。我的mqtt服务器在老家，免费的花生壳流量让我用完了，只能下个月在测试了。

terrance-liang, 2021-02-19:
当前时间最新推送固件版本（目前不在家无法连接插排查看版本号，应该是1.2.0），始终勾选“总是通过UDP发送数据”。间隔一周左右会发生mqtt不更新数据，无法控制，在内网使用app也无法获取任何数据以及控制插排，只能通过插拔插头解决。家里有三个插排都出现过此现象。

terrance-liang, 2021-02-19:
> 当前时间最新推送固件版本（目前不在家无法连接插排查看版本号，应该是 1.2.0），始终勾选 “总是通过 UDP 发送数据”。间隔一周左右会发生 mqtt 不更新数据，无法控制，在内网使用 app 也无法获取任何数据以及控制插排，只能通过插拔插头解决。家里有三个插排都出现过此现象。

查看了自己在另一个issue的留言，之前插排板本分别是1.0.2、0.10.3的时候未出现此现象

kuailezhiyuan, 2021-02-20:
我的经常插排也会出现这个问题，原来的固件版本不会出现

terrance-liang, 2021-02-21:
> > 当前时间最新推送固件版本（目前不在家无法连接插排查看版本号，应该是 1.2.0），始终勾选 “总是通过 UDP 发送数据”。间隔一周左右会发生 mqtt 不更新数据，无法控制，在内网使用 app 也无法获取任何数据以及控制插排，只能通过插拔插头解决。家里有三个插排都出现过此现象。
> 
> 查看了自己在另一个 issue 的留言，之前插排板本分别是 1.0.2、0.10.3 的时候未出现此现象

更新：确认版本号为1.2.0，家里三个插排最近三天已经出现四次此情况，app消息显示掉线的插排“设备离线（功能调试中）"

AgentShanYang, 2021-04-18:
我也遇到了类似情况，过几天TC1就会跟MQTT服务器断连，但是ha和手机app依旧可以读取到插座的功率信息，只是无法操纵开关，唯有插拔电源重启才能解决，两台TC1版本号都是1.2.1，未勾选总是UDP发送数据选项，上报频率是5秒

mzfe, 2021-09-25:
同样有此问题，9天多一点。1.2.1版本，ha和手机app能看到的功率，但无法控制插座。

liyiecho, 2021-10-03:
同样有此问题，1.2.1版本，HA和手机均无法控制插座，切换了mqtt地址也同样连接失败，上一次是10天前，今天又出现了这问题，只能通过插拔电源重启来解决问题

Willy1216, 2023-01-01:
最近遇到这个问题了，是还没有被修复吗

AgentShanYang, 2023-01-01:
> 最近遇到这个问题了，是还没有被修复吗

不会修复了，不定期插拔电源吧

a2633063, 2023-11-03:
查看了下共享的mqtt服务器, ztc1最长时间连接到现在已经6个月左右,   随便看了3个版本为 1.0.2 1.2.0

waylon-w, 2025-01-31:
一般三天左右就会出现，这是个致命BUG

xe5700, 2026-02-21:
这个问题我也遇到过，但是要很久很久，开三个月以上有可能会遇到，之前用的v1.2.0固件，v1.3.0还没试过有没有这种情况发生。