a2633063/zTC1 #64: 尝试修改了一下接入hass的代码 避免报错

BillionReyz, 2020-03-07:
> 请先搜索确认没有重复后再提交issue,以下内容若未按照模板填写,不做任何回复,3天后若未更新直接关闭issue.

固件版本(必填): 1.02

App版本(与app控制有关时必填):

Android版本(与app控制有关时必填):

是否使用服务器(必填): MQTT

问题描述(包含复现issue方式):

问题出在执行payload的时候必须为string格式而不能是json，自己稍微修改了一下

switch:
  - platform: mqtt
    name: "ztc1_1_MACMAC"
    state_topic: "device/ztc1/MACMAC/state"
    command_topic: "device/ztc1/MACMAC/set"
    payload_on: "{\"mac\":\"MACMAC\",\"plug_0\":{\"on\":1}}"
    payload_off: "{\"mac\":\"MACMAC\",\"plug_0\":{\"on\":0}}"
    value_template: "{{ value_json.plug_0.on }}"
    state_on: "1"
    state_off: "0"
  - platform: mqtt
    name: "ztc1_2_MACMAC"
    state_topic: "device/ztc1/MACMAC/state"
    command_topic: "device/ztc1/MACMAC/set"
    payload_on: "{\"mac\":\"MACMAC\",\"plug_1\":{\"on\":1}}"
    payload_off: "{\"mac\":\"MACMAC\",\"plug_1\":{\"on\":0}}"
    value_template: "{{ value_json.plug_1.on }}"
    state_on: "1"
    state_off: "0"
  - platform: mqtt
    name: "ztc1_3_MACMAC"
    state_topic: "device/ztc1/MACMAC/state"
    command_topic: "device/ztc1/MACMAC/set"
    payload_on: "{\"mac\":\"MACMAC\",\"plug_2\":{\"on\":1}}"
    payload_off: "{\"mac\":\"MACMAC\",\"plug_2\":{\"on\":0}}"
    value_template: "{{ value_json.plug_2.on }}"
    state_on: "1"
    state_off: "0"
  - platform: mqtt
    name: "ztc1_4_MACMAC"
    state_topic: "device/ztc1/MACMAC/state"
    command_topic: "device/ztc1/MACMAC/set"
    payload_on: "{\"mac\":\"MACMAC\",\"plug_3\":{\"on\":1}}"
    payload_off: "{\"mac\":\"MACMAC\",\"plug_3\":{\"on\":0}}"
    value_template: "{{ value_json.plug_3.on }}"
    state_on: "1"
    state_off: "0"
  - platform: mqtt
    name: "ztc1_5_MACMAC"
    state_topic: "device/ztc1/MACMAC/state"
    command_topic: "device/ztc1/MACMAC/set"
    payload_on: "{\"mac\":\"MACMAC\",\"plug_4\":{\"on\":1}}"
    payload_off: "{\"mac\":\"MACMAC\",\"plug_4\":{\"on\":0}}"
    value_template: "{{ value_json.plug_4.on }}"
    state_on: "1"
    state_off: "0"
  - platform: mqtt
    name: "ztc1_6_MACMAC"
    state_topic: "device/ztc1/MACMAC/state"
    command_topic: "device/ztc1/MACMAC/set"
    payload_on: "{\"mac\":\"MACMAC\",\"plug_5\":{\"on\":1}}"
    payload_off: "{\"mac\":\"MACMAC\",\"plug_5\":{\"on\":0}}"
    value_template: "{{ value_json.plug_5.on }}"
    state_on: "1"
    state_off: "0"

  - platform: mqtt
    name: "ztc1_power_MACMAC"
    state_topic: "device/ztc1/MACMAC/sensor"
    unit_of_measurement: 'W'
    icon: mdi:gauge
    value_template: "{{ value_json.power }}"
  - platform: mqtt
    name: "ztc1_time_MACMAC"
    state_topic: "device/ztc1/MACMAC/sensor"
    #unit_of_measurement: '秒'
    icon: mdi:gauge
    #value_template: '{{ value_json.total_time }}'
    value_template: >-
      {% set time = value_json.total_time %}
      {% set minutes = ((time % 3600) / 60) | int %}
      {% set hours = ((time % 86400) / 3600) | int %}
      {% set days = (time / 86400) | int %}
      {%- if time < 60 -%}
        <1分钟
      {%- else -%}
        {%- if days > 0 -%}
            {{ days }}天
        {%- endif -%}
        {%- if hours > 0 -%}
            {{ hours }}小时
        {%- endif -%}
        {%- if minutes > 0 -%}
            {{ minutes }}分钟
        {%- endif -%}
      {%- endif -%}
    
homeassistant:
  customize:
    switch.ztc1_1_MACMAC:
      friendly_name: zTC1插槽1
    switch.ztc1_2_MACMAC:
      friendly_name: zTC1插槽2
    switch.ztc1_3_MACMAC:
      friendly_name: zTC1插槽3
    switch.ztc1_4_MACMAC:
      friendly_name: zTC1插槽4
    switch.ztc1_5_MACMAC:
      friendly_name: zTC1插槽5
    switch.ztc1_6_MACMAC:
      friendly_name: zTC1插槽6
    sensor.ztc1_power_MACMAC:
      friendly_name: zTC1功率
    sensor.ztc1_time_MACMAC:
      friendly_name: zTC1运行时间

a2633063, 2020-03-07:
抱歉 除了单引号换为双引号没看出差别  据我测试 单双引号应该没有区别
有其他修改的地方吗

fantasyjqc, 2020-03-07:
我在用node-red接入的是也遇到这样的情况，表现为，如果直接在node-red的mqtt节点中输出解析的JSON就会报错，错误提示是json格式错误，但是如果输出string，然后自己使用JSON.parse(string)可以正常解析，不知道问题在哪里

a2633063, 2020-03-07:
试了一下  并没有报错啊?  提供下相关配置看看

BillionReyz, 2020-03-07:
哎？我可能复制错了 我在类似于payload后面的引号前都加了\的通配符，类似于
payload_on: "{\"mac\":\"MACMAC\",\"plug_0\":{\"on\":1}}"
payload_off: "{\"mac\":\"MACMAC\",\"plug_0\":{\"on\":0}}"

a2633063, 2020-07-19:
经确认,原写法不会报错.
请确认其他原因

closed 2020-07-19 by a2633063