a2633063/zTC1 #67: 功率上报持续为-1W

Freedomzasf, 2020-07-02:
> 请先搜索确认没有重复后再提交issue,以下内容若未按照模板填写,不做任何回复,3天后若未更新直接关闭issue.

固件版本(必填):
1.1.0
App版本(与app控制有关时必填):
1.4.4
Android版本(与app控制有关时必填):
10.0
是否使用服务器(必填):
无服务器
问题描述(包含复现issue方式):
功率显示一直为-1W，其余功能正常，重启、重新配网等均无效

picc745514, 2020-07-03:
同样问题，有一次开机5天，出现功率了，重启又变成-1W了

conscloud, 2020-07-19:
同样的问题，我有两个，有一个用着用着突然功能变成-1了，另外一个原来一起是-1突然又有功率了。

kevinvcon, 2020-08-15:
今天我的TC1也是显示-1的功率了  这是什么问题呢？

shineanr, 2020-09-07:
[image]

我发现再开机插槽的时候有几率会出现功率值

a2633063, 2020-09-20:
可能为硬件问题。固件不确认能否处理。由于无强电硬件调试环境，暂时不处理此问题
APP显示-1是为表示当前功率异常
此问题在设备0TA/重启时可能恢复或出现

reg1717, 2020-09-26:
[image]

刷了1.1.0以后就没怎么管它，今天点开ha发现功率-1了

a2633063, 2020-09-26:
硬件问题, 概率性出现,也可能概率性恢复. 更换功率ic可完全恢复

reg1717, 2020-10-11:
好的，不影响使用就不管了，感谢您的固件！

closed 2020-11-07 by a2633063

fengsuxing, 2021-01-11:
遇到了这问题，量功率芯片电压正常，，换功率芯片解决。淘宝几块钱能买好几个

walker365, 2021-04-30:
> 遇到了这问题，量功率芯片电压正常，，换功率芯片解决。淘宝几块钱能买好几个

有型号或者关键词吗，我也出现这个问题了，我也想自己把他换了

walker365, 2021-04-30:
对于上面的回复追加一些信息，这是在恩山上看到的功率不显示的问题，里面有具体元件型号和可能损坏的元件，提供参考，我自己也遇到了功率-1的情况，我自己近期也会更换
https://www.right.com.cn/forum/thread-951885-1-1.html

shineanr, 2022-04-09:
更换芯片后半年又出现功率-1，不知道是否跟使用usb口充电有关，我另一个基本不使用usb口充电，就没有任何问题。

fengsuxing, 2022-08-09:
> 更换芯片后半年又出现功率-1，不知道是否跟使用usb口充电有关，我另一个基本不使用usb口充电，就没有任何问题。

估计是电路设计有问题，我的也又坏了，现在懒的修了

moodnood, 2025-04-18:
路由器升级wifi7了，这个tc1总是断连。有办法解决吗。tp 7260路由器。设置bgn也不好使。

moodnood, 2025-04-18:
路由器升级wifi7了，这个tc1总是断连。有办法解决吗。tp 7260路由器。设置bgn也不好使。

用wif6路由器单开一个wifi正常使用