a2633063/zTC1 #71: ztc1固件支持ipv6的mqtt地址吗？

stone2289, 2021-01-11:
> 请先搜索确认没有重复后再提交issue,以下内容若未按照模板填写,不做任何回复,3天后若未更新直接关闭issue.

固件版本(必填):v1.2.0

App版本(与app控制有关时必填):v1.10.1

Android版本(与app控制有关时必填):安卓10

是否使用服务器(必填):是

问题描述(包含复现issue方式):之前在qq上听您说过ztc1支持ipv6的mqtt服务器，但是我一直没有实现。我用的是mosquitto，网络环境：有ipv6地址的移动宽带，利用docker版的sanjusss/aliyun-ddns实现阿里云的动态域名解析，ztc1所在的WiFi有ipv6。APP可以正常连接mqtt服务器，点击同步mqtt后可以显示已设置mqtt信息。重启app后左下角日志提示app已连接mqtt服务器，当前发送消息：mqtt，设备离线，需要同步mqtt才能连接。另外，相同的网络环境下用花生壳的内网穿透得到的mqtt地址（ipv4），ztc1可以正常的连接。我相关知识都是自学的，可能有表达不清楚的地方请见谅。

aasqaasq1, 2021-01-22:
我也是移动宽带的IPV6 用的阿里云DDNS  
你可以用局域网192开头,MQTT 连接一次 然后再改用IPV6 的域名, 之后就可以使用IPV6的外网连接了,,
但是一般使用一段时间 1个月 左右 会断开链接..需要重启插座.再重复上面的 又可以用..目前只能是这样子

fangtao331, 2025-02-28:
> 我也是移动宽带的IPV6 用的阿里云DDNS  
> 你可以用局域网192开头,MQTT 连接一次 然后再改用IPV6 的域名, 之后就可以使用IPV6的外网连接了,,
> 但是一般使用一段时间 1个月 左右 会断开链接..需要重启插座.再重复上面的 又可以用..目前只能是这样子

真是太感谢了!搞了一天都不知道什么原因。