a2633063/zTC1 #79: 只要涉及到断开WIFI的情况，插排就会失连，无法控制

shinesky, 2022-07-09:
感谢作者的固件，不知道作者还维护否，看到去年有相同的issue
同样的问题，如果路由器重启，或者插排重启，断电等操作，都会导致wifi断开，wifi断开后插排就连不上了，无法控制，只能重新配网。

固件版本(必填):1.2.1

App版本(与app控制有关时必填):1.11.3

Android版本(与app控制有关时必填):11

是否使用服务器(必填):使用mqtt

问题描述(包含复现issue方式):插排正常使用的情况下，一旦重启路由器，或者插排断电重启，这些操作都会导致wifi断开，插排就再也连不上wifi，指示灯一直快速闪烁，只能重新配网设置。

a2633063, 2022-07-09:
并没有复测到此问题 
我使用4台 都没有出现过此问题 路由器重启都能正常连接 更换路由器使用相同ssid和密码也能及时连接上

shinesky, 2022-07-09:
> 并没有复测到此问题 我使用4台 都没有出现过此问题 路由器重启都能正常连接 更换路由器使用相同ssid和密码也能及时连接上

刚刚我再一次尝试，走过去直接拔了它的电源，过了一分钟插回去，通电后指示灯开始狂闪，我一直等待了5分钟，指示灯还是在一直狂闪，也没有连上wifi，无奈只能重新配网。😥
看到这个也是同样的情况，wifi断开后设备失联
https://github.com/a2633063/zTC1/issues/74

zzzop, 2022-08-14:
我这里灯不闪，但是重启路由器或者插排APP就连不上插排

laojimigu, 2025-09-02:
遇到好多次相同情况，只要路由器断网就有几率发生。不过我这儿看起来大概是dhcp方面的问题，TC1不知道自己租期到了，所以不会续约也不会重新分配。
我每次都是重新配网解决的，重新插拔TC1电源也可以解决。准备测试一下不手动处理 等内部维护的租约到期能不能自行恢复

cnwelee, 2026-09-15:
我这也有类似问题，待会试试绑定MAC地址固定IP看看

cnwelee, 2026-09-15:
> 我这也有类似问题，待会试试绑定MAC地址固定IP看看

绑定MAC地址好像不管用，我这一会儿就失联了，都没等到dhcp租约过期……